W0rthlessS0ul/nRF24_jammer #2: error

samukas81, 2024-10-24:
Hello, how do I modify it to 128x64 OLED?
I modified it in Sketchup and recompiled it in .bin but it didn't work

W0rthlessS0ul, 2024-10-25:
Hello,

The display of images on the screen is achieved by modifying bitmap images. All images are currently set for a resolution of 128x32. In this message, I have attached a ZIP archive named jpg.zip containing .jpg images. Please resize them to 128x64 and update the entries in bitmap.h accordingly. 
[jpg.zip](https://github.com/user-attachments/files/17515840/jpg.zip)

samukas81, 2024-10-25:
wow tnk's!!!

W0rthlessS0ul, 2024-10-25:
I was glad to help! If it’s not too much trouble, I would appreciate it if you could add my repository to your favorites. Doing so would greatly assist with its promotion. Thank you!

samukas81, 2024-10-25:
Yes I will add it, and congratulations, it works very well!

W0rthlessS0ul, 2024-10-25:
I'm glad to hear that! Wishing you successful usage.

samukas81, 2024-10-25:
sorry, is bitmap.h

W0rthlessS0ul, 2024-10-25:
Alright, I can modify the firmware for the 128x64 display. However, I don’t have that type of display on hand, so I won't be able to test it myself. After you update the firmware, I would appreciate it if you could send me a photo of the device. I'll do my best to send the modified firmware as soon as possible, probably within 1-2 hours, though it could be a bit earlier or later.

W0rthlessS0ul, 2024-10-25:
How would you prefer me to send the files: as .ino or .bin?

samukas81, 2024-10-25:
.ino please, thank you.

W0rthlessS0ul, 2024-10-25:
Could you please install this firmware and send a photo of what is displayed on your screen? Thank you!
[nRF24_jammer.zip](https://github.com/user-attachments/files/17522411/nRF24_jammer.zip)

samukas81, 2024-10-25:
show oled screen?

W0rthlessS0ul, 2024-10-25:
ya

samukas81, 2024-10-25:
It didn't work, I'm going back to the previous mode where I only changed the 128x64 of the OLED

https://github.com/user-attachments/assets/f07cc49a-3672-4453-8d08-07f277b807a3

W0rthlessS0ul, 2024-10-25:
Could you please clarify if the display showed anything on the previous firmware version?

samukas81, 2024-10-25:
Yes, it did, taking advantage of the subject, is there a way to modify it so that you don't need to turn off the power to be able to choose a new BLE, WiFi, etc. method?

samukas81, 2024-10-25:
[image]

W0rthlessS0ul, 2024-10-25:
Oh, I'm sorry! I thought your display wasn't working at all. In that case, please try this firmware.
[nRF24_jammer.zip](https://github.com/user-attachments/files/17522860/nRF24_jammer.zip)

W0rthlessS0ul, 2024-10-25:
Unfortunately, the display and the NRF24 module are connected to the same pins. As a result, the display can function initially while the NRF24 is not yet initialized. However, once the NRF24 is activated, the display will no longer work. Consequently, after the attack is launched, the only way to regain functionality is to reboot the device.

samukas81, 2024-10-25:
[image]

W0rthlessS0ul, 2024-10-25:
The image quality isn't the best, but at least it's functional. Thank you for helping me understand how the 128x64 display works. In the next update, I will try to add support for the 128x64 display. However, simply stretching the images isn't a viable solution.

If this option doesn't meet your needs, you can modify the imag
es to fit your resolution using the attached files, which are available in both .jpg and .psd formats. To convert the .jpg files into bitmap images, you can use the following tool: [img2cpp](https://javl.github.io/image2cpp/)
[img.zip](https://github.com/user-attachments/files/17522994/img.zip)

samukas81, 2024-10-25:
Thank you very much my friend to help

W0rthlessS0ul, 2024-10-25:
I believe we can consider this issue resolved—would you agree?

samukas81, 2024-10-25:
Yes!

W0rthlessS0ul, 2024-10-25:
Wishing you good luck, and goodbye for now!

closed 2024-10-25 by W0rthlessS0ul