W0rthlessS0ul/nRF24_jammer #6: Copyright Infridgement

EmenstaNougat, 2024-11-14:
Copied parts of repository such as pinouts and inspiration/idea without credits.

https://github.com/EmenstaNougat/ESP32-BlueJammer

[image]

1:1 GPIO pinout as my ESP32-BlueJammer? This can't be a coincidence ;)

Jamming drones? Referred to my RC?

[image]

W0rthlessS0ul, 2024-11-14:
Hello, I’m afraid I was inspired not by your code, but rather by the code from this repository: https://github.com/smoochiee/Bluetooth-jammer-esp32.

W0rthlessS0ul, 2024-11-14:
By the way, thank you for reminding me to credit the author; I'll make sure to cite smoochiee right away.

EmenstaNougat, 2024-11-14:
Yeah, but I don't see my GPIO pinouts anywhere on smoochies repo

W0rthlessS0ul, 2024-11-14:
I could have credited you in the Acknowledgments section if I had taken a part of your code. However, I cannot include you simply for the similar tables we created.

W0rthlessS0ul, 2024-11-16:
Can we consider this issue resolved?

EmenstaNougat, 2024-11-16:
Sure

closed 2024-11-16 by W0rthlessS0ul

smoochiee, 2024-11-28:
> I could have credited you in the Acknowledgments section if I had taken a part of your code. However, I cannot include you simply for the similar tables we created.

lol...why he think my repo dont have gpio pin out..🤔when its clearly there and he built my work before he started his...😂.shame

W0rthlessS0ul, 2024-11-28:
Who knew GPIO pinouts could spark such intense discussions? 😂 Next thing you know, we’ll be debating who invented the wheel!

Since Smoochiee released his work on March 6, 2024, and you posted yours later on July 7, 2024, and considering that your repository isn't open source, I can assert that you used Smoochiee's code when creating your repository. 

So, I hope this clarifies things, and we won’t need to revisit this topic again!

EmenstaNougat, 2024-11-28:
I'm not about the jammer stuff

W0rthlessS0ul, 2026-10-07:
Funny how life works. A while ago, this issue was opened titled **"Copyright Infridgement"** to lecture me about hardware GPIO pinouts

Fast forward to today: static analysis of that closed firmware binary revealed that the only actual copyright infringement taking place was the statically linked, unlicensed use of the **GPLv2** [RF24 library](https://github.com/nRF24/RF24)

When this was documented with full decompilation evidence, @EmenstaNougat promptly panicked and wiped the issue to cover the tracks

Fortunately, you don’t have moderator privileges on this repository to delete inconvenient facts, and the full technical breakdown is already permanently preserved for anyone interested in upstream license compliance:
👉 [https://archive.ph/RIKUr](https://archive.ph/RIKUr)

Attempting to erase evidence after being caught doesn't resolve upstream licensing violations - it only confirms willful infringement. What happens next with the upstream code and maintainers is out of your hands **;P**