W0rthlessS0ul/nRF24_jammer #9: Wifi channel

samukas81, 2024-11-30:
Could you implement specific wifi channel selection to deauthenticate?

W0rthlessS0ul, 2024-11-30:
Hello, that's an interesting idea! I'll do my best to implement it in the next update.

samukas81, 2024-11-30:
Doing more tests, I discovered that in all modes after reset it still sends some kind of signal. Only when it turns off does it stop, but in Bluetooth mode after reset it stops the interference. Sorry for the video quality, I had to reduce the size to be able to post.

https://github.com/user-attachments/assets/ec3fee8d-7ae8-421b-9104-4eff3395272c

W0rthlessS0ul, 2024-11-30:
I'm not sure why, but the video you sent is opening as an MP3 file. Could you please create a repository and upload your video there so that I can watch it?

samukas81, 2024-11-30:
download the vídeo

samukas81, 2024-11-30:
Did you manage to watch the video?

W0rthlessS0ul, 2024-11-30:
I'm a little busy right now, sorry I can't take a look.

samukas81, 2024-11-30:
Don't worry, if you can't see it, let me know later, hugs.

W0rthlessS0ul, 2024-11-30:
I have a little bit of free time, but I might be unavailable again soon. I’m having trouble downloading your video. If it's not too much trouble, could you please send it to my email: [w0rthlesss00ul@gmail.com](mailto:w0rthlesss00ul@gmail.com)? Thank you!

W0rthlessS0ul, 2024-11-30:
An access point is being created on the ESP32, which also operates within the 2400 MHz to 2500 MHz range. However, my device does not transmit that much data, so either you are using an ESP32 with a more powerful transmitter (though that's unlikely), or there are issues with connecting the NRF24. It is definitely not a software issue, as the NRF24 is initialized only after the attack is launched.

samukas81, 2024-11-30:
How do I disable this access point that it creates, which lines do I disable in the sketch?

W0rthlessS0ul, 2024-11-30:
Please insert the following code into the nRF24_jammer.ino file:
#include "SPI.h"
#include "Update.h"
#include "EEPROM.h"
#include "Adafruit_GFX.h"
#include "Adafruit_SSD1306.h"
#include "GyverButton.h"
#include "RF24.h"
#include "bitmap.h"
#include "html.h"
#include "WiFi.h"
#include "WebServer.h"
#include "options.h"
#include "jam.h"

void handleRoot() {
    server.send(200, "text/html", html);
}

void sendHtmlAndExecute(const char* htmlResponse, void (*action)() = nullptr) {
    server.send(200, "text/html", htmlResponse);
    if (action) action();
}

void jamHandler(const char* htmlResponse, void (*jamFunction)()) {
    sendHtmlAndExecute(htmlResponse, jamFunction);
}

void miscChannelsHandler() {
    int channel1 = server.arg("start").toInt();
    int channel2 = server.arg("stop").toInt();
    sendHtmlAndExecute(html_misc_jam);
    misc_jam(channel1, channel2);
}

void settingsHandler(const char* htmlResponse) {
    sendHtmlAndExecute(htmlResponse);
}

void storeEEPROMAndSet(int index, int value, int& targetVar) {
    EEPROM.write(index, value);
    EEPROM.commit();
    targetVar = value;
    sendHtmlAndExecute(html);
}

void registerRoute(const char* path, void (*handler)()) {
    server.on(path, handler);
}

void handleFileUpload() {
    HTTPUpload &upload = server.upload();
    static unsigned long lastUpdate = 0; 
    const unsigned long updateInterval = 100;

    if (upload.status == UPLOAD_FILE_START) {
        Serial.printf("Update: %s\n", upload.filename.c_str());
        if (!Update.begin(UPDATE_SIZE_UNKNOWN)) {
            Update.printError(Serial);
            server.send(200, "text/plain", "Update Failed");
            return;
        } else {
            server.sendContent(html_upload_progress);
            lastUpdate = millis();
        }
    } 
    else if (upload.status == UPLOAD_FILE_WRITE) {
        if (Update.write(upload.buf, upload.currentSize) != upload.currentSize) {
            Update.printError(Serial);
        }
[... 123 lines cut ...]
        display.drawBitmap(0, 0, bitmap, 128, 64, WHITE);
        display.display();
    }
    
    if (butt1.isHolded()) {
        if (menu_number == 5){
          misc();
        }
        display.clearDisplay();
        const uint8_t* bitmap = (menu_number == 0) ? bitmap_bluetooth_jam :
                                (menu_number == 1) ? bitmap_drone_jam :
                                (menu_number == 2) ? bitmap_wifi_jam : 
                                (menu_number == 3) ? bitmap_ble_jam : bitmap_zigbee_jam;
        display.drawBitmap(0, 0, bitmap, 128, 64, WHITE);
        display.display();
        
        switch (menu_number) {
            case 0: bluetooth_jam(); break;
            case 1: drone_jam(); break;
            case 2: wifi_jam(); break;
            case 3: ble_jam(); break;
            case 4: zigbee_jam(); break;
        }
    }
}

W0rthlessS0ul, 2024-11-30:
Please insert the following code into the nRF24_jammer.ino file:
```#include "SPI.h"
#include "Update.h"
#include "EEPROM.h"
#include "Adafruit_GFX.h"
#include "Adafruit_SSD1306.h"
#include "GyverButton.h"
#include "RF24.h"
#include "bitmap.h"
#include "html.h"
#include "WiFi.h"
#include "WebServer.h"
#include "options.h"
#include "jam.h"

void handleRoot() {
    server.send(200, "text/html", html);
}

void sendHtmlAndExecute(const char* htmlResponse, void (*action)() = nullptr) {
    server.send(200, "text/html", htmlResponse);
    if (action) action();
}

void jamHandler(const char* htmlResponse, void (*jamFunction)()) {
    sendHtmlAndExecute(htmlResponse, jamFunction);
}

void miscChannelsHandler() {
    int channel1 = server.arg("start").toInt();
    int channel2 = server.arg("stop").toInt();
    sendHtmlAndExecute(html_misc_jam);
    misc_jam(channel1, channel2);
}

void settingsHandler(const char* htmlResponse) {
    sendHtmlAndExecute(htmlResponse);
}

void storeEEPROMAndSet(int index, int value, int& targetVar) {
    EEPROM.write(index, value);
    EEPROM.commit();
    targetVar = value;
    sendHtmlAndExecute(html);
}

void registerRoute(const char* path, void (*handler)()) {
    server.on(path, handler);
}

void handleFileUpload() {
    HTTPUpload &upload = server.upload();
    static unsigned long lastUpdate = 0; 
    const unsigned long updateInterval = 100;

    if (upload.status == UPLOAD_FILE_START) {
        Serial.printf("Update: %s\n", upload.filename.c_str());
        if (!Update.begin(UPDATE_SIZE_UNKNOWN)) {
            Update.printError(Serial);
            server.send(200, "text/plain", "Update Failed");
            return;
        } else {
            server.sendContent(html_upload_progress);
            lastUpdate = millis();
        }
    } 
    else if (upload.status == UPLOAD_FILE_WRITE) {
        if (Update.write(upload.buf, upload.currentSize) != upload.currentSize) {
            Update.printError(Serial);
        }
[... 123 lines cut ...]
        display.drawBitmap(0, 0, bitmap, 128, 64, WHITE);
        display.display();
    }
    
    if (butt1.isHolded()) {
        if (menu_number == 5){
          misc();
        }
        display.clearDisplay();
        const uint8_t* bitmap = (menu_number == 0) ? bitmap_bluetooth_jam :
                                (menu_number == 1) ? bitmap_drone_jam :
                                (menu_number == 2) ? bitmap_wifi_jam : 
                                (menu_number == 3) ? bitmap_ble_jam : bitmap_zigbee_jam;
        display.drawBitmap(0, 0, bitmap, 128, 64, WHITE);
        display.display();
        
        switch (menu_number) {
            case 0: bluetooth_jam(); break;
            case 1: drone_jam(); break;
            case 2: wifi_jam(); break;
            case 3: ble_jam(); break;
            case 4: zigbee_jam(); break;
        }
    }
}```

samukas81, 2024-11-30:
Does not compile

W0rthlessS0ul, 2024-12-01:
[nRF24_jammer.zip](https://github.com/user-attachments/files/17968532/nRF24_jammer.zip)
Install all the .bin files from the ZIP archive above onto your ESP32.

W0rthlessS0ul, 2024-12-09:
> Could you implement specific wifi channel selection to deauthenticate?

Your idea has been successfully implemented, so we can consider the issue resolved. Do you agree?

samukas81, 2024-12-09:
yes

W0rthlessS0ul, 2024-12-09:
Good luck! If you have any more ideas to contribute, feel free to create additional issues and share your suggestions there.

closed 2024-12-09 by W0rthlessS0ul

samukas81, 2024-12-09:
Congratulations on your commitment!!

W0rthlessS0ul, 2024-12-09:
Thank you! I appreciate your support and feedback. It motivates me to keep improving the project!