W0rthlessS0ul/nRF24_jammer #14: Add more buttons

shuixing1122, 2024-12-26:
https://github.com/user-attachments/assets/e9f2d285-0a6e-4bd0-b124-eb16e9ed7a49
Hello, could you please add more buttons to control? Only one button is too troublesome to control. Just like that video, there are four buttons, up and down, OK and return. It's very convenient to operate like this.

W0rthlessS0ul, 2024-12-30:
Hello, this is a great idea! However, in order to update the firmware without errors, I need to create another device. Unfortunately, I currently have very limited funds, so I will have to attempt this later when my financial situation improves. 
But thank you for the great idea!

shuixing1122, 2024-12-31:
I saw on other forums that someone else adopted your firmware and also made a PCB board, the effect looks really cool, looking forward to your new firmware😊

https://github.com/user-attachments/assets/1c7d1b52-f10f-476f-9bea-cb57e9d3245d

W0rthlessS0ul, 2025-02-06:
Hello again! Sorry for the delay in releasing updates. I had some issues that prevented me from working on GitHub for a while, but I'm excited to share that your suggestion has been included in the latest update!

W0rthlessS0ul, 2025-02-07:
Can we consider this issue resolved?

closed 2025-02-07 by shuixing1122