W0rthlessS0ul/nRF24_jammer #17: Suggestion (notice me please 🥺🥺🥺)

ilsaf4ik, 2025-04-01:
Could you add cc1101 jammer? Its a really cool thing

W0rthlessS0ul, 2025-04-01:
Hello! It’s unlikely that we can add the CC1101 to this project, as I’d prefer not to mix everything together. However, I could create a separate project specifically dedicated to the CC1101 jammer.

Would this option work for you?

Abhishek2142x, 2025-04-04:
Hello dev sir,  can you guide me to link for  Ble spam/beacon spamer using nrf 52840 please :)

cYBERAPELSINn, 2025-04-05:
Wow, it would be cool if you made a cc1101 jammer because i dont understand how does it work

Michael-1357, 2025-09-20:
Hi,  very impressive.  I'm mostly interested in the potential for drone jamming.

Given the radios also receive have you considered adding drone detection?   To passively detect, monitor, and log detections and send alerts via the web interface? 

Perhaps also the option to begin jamming appropriate channels automatically when a drone is detected and then continue for a configurable amount of time before returning to scanning for drones?  

Would prolong battery life, be a tad more stealthy, and perhaps less collateral interference for other radio spectrum users.  As far as I can see it shouldn't require any hardware modification or additional components.