W0rthlessS0ul/nRF24_jammer #18: Can you elaborate the process for inputing code, soldering, etc for noobies like me?(maybe like a vid or smth)

MaximusMan11, 2025-04-02:
Basically, what the title says^

W0rthlessS0ul, 2025-04-05:
Hello! I will add a video about flashing the firmware, but I probably won't be able to add a video on soldering.

dr1408, 2026-01-20:
I am also noob I bought all parts but dunno how to solder and build

djmuerteitm, 2026-01-21:
Hi, dr!

You can begin without soldering if you use a prototype breadboard and some dupont wires... No solder needed...

djmuerteitm, 2026-01-21:
Depending nrf modules, must be wired at 5v or 3.3v. 2g4m27d supports 5v directly, but "normal" nrf24 runs on 3.3 without adapter module. If you use adapter, feed It (the module) with 5v.

Obviously, 2g4m27d modules are so far better than shitty nrf24 others... 💩

Your module fits perfect for this setup, but i hope that you have added too many stain on pin headers... I prefer build this stuff on a double sided "blank" PCB, and solder pin female sockets for each module. Easy to attach, dettach or re-use mcus and modules without overheating It by solder-desolder... Check the pictures of my setups, that i have been sent on posts above this...

In both setups, pay attention on use decent wires (do not use shitty cables like cheap chinese earphones, i mean), do not blow to soldering points for quick cold down them, and work like a brain surgeon if you can 🤣