W0rthlessS0ul/nRF24_jammer #22: Low range

narrwass, 2025-04-20:
How can i extend the range of the bluetooth jammer?
Im connecting my phone with my car stereo and the jammer needs to be in the range of 2 meters to stop the music.

Ive connected the 3.3V to the esp 3.3V pin.Maybe use external power supply? Didnt measured the voltage drop but the display is powered by the same pin and its ok.
Using the one anthena module version with jammer_C_32 code.

W0rthlessS0ul, 2025-04-20:
Hello, you can buy a 2.4 Ghz amplifier and connect the jammer to the RFin input and the antenna to the RFout (for example, I use a 1 Watt TZT RF2126).

You can also use a directional antenna, it also greatly increases the range of operation (I use NOWEPOCH 2.4G WIFI at 18dBi).

djmuerteitm, 2025-07-16:
You can use Ebyte 2G4M27 modules instead "basic" nrf24. Ebyte can tx Up to 27db, and with standard build setup (2 modules) and proper antennas you can extend the range above 10 or 20 meters for easy (outdoors)