W0rthlessS0ul/nRF24_jammer #26: Freeze

samukas81, 2025-05-22:
Hello friend, when I press the button to stop the attack it stops, but the screen freezes and nothing else works.

W0rthlessS0ul, 2025-05-23:
Hello, to help, I need to know:

1. Which release do you use
2. When exiting from which attack (bluetooth, drone, ble or something else), the jam occurred
3. If possible, could you describe in detail what steps you took before the error occurred so that I can reproduce it?

samukas81, 2025-05-23:
I tested with the latest and penultimate version, same problem. The problem occurs with any attack method. I installed it both via OTA and directly via Arduino V2.

samukas81, 2025-05-25:
Another problem I found was that even after stopping the attack, the modules remain activated with tree

[image]

 carriers in the band. It only turns off by removing the power supply.

W0rthlessS0ul, 2025-05-26:
The problem with the module remaining active has been fixed in release V2.5.0 
Regarding the issue with hang, I was unable to replicate this error. Could you please send [me](mailto:w0rthlesss00ul@gmail.com) a video demonstrating what you are doing and why the issue is occurring?

W0rthlessS0ul, 2025-06-01:
Hello again, I understand what you were talking about, I will try to fix it soon.

samukas81, 2025-06-03:
Hello again, now it's fine but it still has active carriers after canceling the attack.

samukas81, 2025-06-03:
In all!!

W0rthlessS0ul, 2025-06-03:
Okay, I'll see if it can be fixed

samukas81, 2025-06-15:
Hello, did you manage to fix it?

W0rthlessS0ul, 2025-06-16:
Hello, I'm afraid I haven't been able to find this problem, here's what my spectrum looks like:

[image]

As you can see, after stopping the jammer, the spectrum remains the same as it was before