W0rthlessS0ul/nRF24_jammer #29: Where are the schematics for the required HW?

eabase, 2025-05-30:
Hi,
Very cool project!

However, I'm missing the schematics (or links to) for the required hardware. 
I have some ESP32 pieces and some nRF52840 stuff. Can they be used? 

I'm working on a Deauth for the ESP...

eabase, 2025-06-01:
> Did you even read the readme? Very first page.

I did, and I don't see a BOM nor a board or Schematic on how you connected the devices together. As I'm looking to try to use my own devices, that would be very helpful.

If you don't have it, or can't explain it, maybe you can provide some more closeup photos of the breadboard you're using?

W0rthlessS0ul, 2025-06-01:
Hello, in README.md there is a section called "🧑🔧 Let's Get Started with Soldering!" in which the "With OLED" and "Without OLED" drop-down windows select the version you want to assemble or you can watch it below:

## 🧑‍🔧 Let's Get Started with Soldering!
<details>
<summary><strong>With OLED</strong></summary>

<div style="margin-left: 20px;">

## Differences between versions

The **Compact** version is equipped with a **single NRF24** module, while the **Standard** version features **two**.
 
Notably, the Compact version allows **uninterrupted access to the display** even when jamming is started. 

This will enable me to utilize features that may be added in the future but are currently **unavailable** in the **Standard version**. For instance, **one already implemented** feature is the a**bility to exit jamming mode** by simply pressing the "**OK**" button (*pin 25*). 

Given these advantages, **I highly recommend choosing the Compact version** for its versatility and potential for future enhancements.

---

<details>
<summary><strong>Compact</strong></summary>

<div style="margin-left: 20px;">

### HSPI Connection
| **Pin Name** | **ESP32 GPIO** | **Connection**       |
|--------------|----------------|----------------------|
| VCC          | 3.3V          | (+) capacitor        |
| GND          | GND           | (-) capacitor        |
| CE           | GPIO 16       |                      |
| CSN          | GPIO 15       |                      |
| SCK          | GPIO 14       |                      |
| MOSI         | GPIO 13       |                      |
| MISO         | GPIO 12       |                      |
| IRQ          |                |                      |

### OLED Connection
| **Pin Name** | **ESP32 GPIO** |
|--------------|----------------|
| VCC          | 3.3V          |
| GND          | GND           |
| SCL          | GPIO 22       |
| SDA          | GPIO 21       |

### Button Connection
| **Button Actions** | **ESP32 GPIO** |
|--------------|----------------|
| OK          | GPIO 25       |
| NEXT (Optional)             | GPIO 26       |
| PREVIOUS (Optional)            | GPIO 27       |

</div>
</details>

<details>
<summary><strong>Standard</strong></summary>

<div style="margin-left: 20px;">

### HSPI Connection
| **Pin Name** | **ESP32 GPIO** | **Connection**       |
|--------------|----------------|----------------------|
| VCC          | 3.3V          | (+) capacitor        |
| GND          | GND           | (-) capacitor        |
| CE           | GPIO 16       |                      |
| CSN          | GPIO 15       |                      |
| SCK          | GPIO 14       |                      |
| MOSI         | GPIO 13       |                      |
| MISO         | GPIO 12       |                      |
| IRQ          |                |                      |

### VSPI Connection
| **Pin Name** | **ESP32 GPIO** | **Connection**       |
|--------------|----------------|----------------------|
| VCC          | 3.3V          | (+) capacitor        |
| GND          | GND           | (-) capacitor        |
| CE           | GPIO 22       |                      |
| CSN          | GPIO 21       |                      |
| SCK          | GPIO 18       |                      |
| MOSI         | GPIO 23       |                      |
| MISO         | GPIO 19       |                      |
| IRQ          |                |                      |

### OLED Connection
| **Pin Name** | **ESP32 GPIO** |
|--------------|----------------|
| VCC          | 3.3V          |
| GND          | GND           |
| SCL          | GPIO 22       |
| SDA          | GPIO 21       |

### Button Connection
| **Button Actions** | **ESP32 GPIO** |
|--------------|----------------|
| OK          | GPIO 25       |
| NEXT (Optional)             | GPIO 26       |
| PREVIOUS (Optional)            | GPIO 27       |

</div>
</details>

</div>
</details>
<details>
<summary><strong>Without OLED</strong></summary>

<div style="margin-left: 20px;">

### HSPI Connection
| **Pin Name** | **ESP32 GPIO** | **Connection**       |
|--------------|----------------|----------------------|
| VCC          | 3.3V          | (+) capacitor        |
| GND          | GND           | (-) capacitor        |
| CE           | GPIO 16       |                      |
| CSN          | GPIO 15       |                      |
| SCK          | GPIO 14       |                      |
| MOSI         | GPIO 13       |                      |
| MISO         | GPIO 12       |                      |
| IRQ          |                |                      |

### VSPI Connection
| **Pin Name** | **ESP32 GPIO** | **Connection**       |
|--------------|----------------|----------------------|
| VCC          | 3.3V          | (+) capacitor        |
| GND          | GND           | (-) capacitor        |
| CE           | GPIO 22       |                      |
| CSN          | GPIO 21       |                      |
| SCK          | GPIO 18       |                      |
| MOSI         | GPIO 23       |                      |
| MISO         | GPIO 19       |                      |
| IRQ          |                |                      |

</div>
</details>

-----

connor33341, 2025-06-02:
I have a kicad schematic, it's a very bare-bones version of the compact jammer. It also "technically" has a bom. The files are in the .zip, so you can edit it, or make it into a pcb, or whatever.
[hardware.zip](https://github.com/user-attachments/files/20543201/hardware.zip)
[image]