W0rthlessS0ul/nRF24_jammer #31: Problem with Button performance

McVillano, 2025-06-01:
Hello,

First of all thank you for this cool project. I have one problem and one question.

The problem when i swich the button mode on the webpage to 3 buttons, they stop working, don't know why:

https://github.com/user-attachments/assets/70257df1-55a6-4515-9f56-e2b07a089bca

Randomly one button works but not consistent.

The question, wich is the logic in the menus for the one button navigation, i feel it would be usefull to add to the README.
I've guessed that it press to go to the next item in menu and long press to enter, but it would be cool to know.
Also i feel like the difference between press and long-press is too little, i don't know if this can be changed, but it would be cool to maybe get like 2-3s press, in mine i feel like i long press without wanting frecuently.

Also i would like to know wich is the stop attack button combination.

https://github.com/user-attachments/assets/b49d0b94-23f0-436b-80ed-71baa31881e3

Video of the one button 

Thanks in advance

eabase, 2025-06-01:
IDK, know, but this sounds like a button *debounce* issue. 
I bet your FW or circuit has no debounce function built-in.

https://www.picotech.com/library/articles/blog/what-is-switch-bounce-how-to-implement-debounce
https://docs.arduino.cc/built-in-examples/digital/Debounce/
https://electronics.stackexchange.com/questions/694167/should-i-debounce-everything

Linking: #21 and #23

McVillano, 2025-06-01:
> IDK, know, but this sounds like a button _debounce_ issue. I bet your FW or circuit has no debounce function built-in. [...]

Thanks for the answer, what you mention is likely happening in the second vídeo, i will try yo add a resistor and a capacitor like the link you sent said or buy new better buttons 

We still don’t know why when i set config mode for 3 buttons non of them work

W0rthlessS0ul, 2025-06-01:
Hello, you're right, I'll try to add a delay and help with control in the near future (thanks for the idea)

But I didn't fully understand what exactly happens when you switch to 3 buttons, you can send a video with exactly what happens.

W0rthlessS0ul, 2025-06-02:
Your idea was added, but I couldn't recreate the error you mentioned.

closed 2025-06-03 by McVillano