W0rthlessS0ul/nRF24_jammer #36: Stuck on connecting the second module nrf24

Dead3ch, 2025-07-24:
Hi, help me I didn't understand, I connected the first NRF24 module, but I got stuck on connecting the second NRF24 module, where should the jumper wire from the second VCC module lead if 3v3 is occupied by the first contact of the NRF24 ?

W0rthlessS0ul, 2025-07-25:
The second wire must also be connected to the same 3.3V pin. To do this, you need to connect the wires, for example, simply solder them or use a terminal

Dead3ch, 2025-07-25:
and what about the GND connection ?

W0rthlessS0ul, 2025-07-25:
If there are two GND pins on the board, then simply connect the wires to different pins, if there is one pin, then connect it in the same way as in the case of the 3.3V pin.

GKDRBQ, 2025-08-05:
Hello, I like what you made very much. I'm using nrf24. It's easy for the interface to crash. It's suspended and can't be used. flipper