W0rthlessS0ul/nRF24_jammer #37: More brain storm - NOT ISSUE

djmuerteitm, 2025-07-27:
Hi again, W0rTh.

I have three M5stickC (non plus, 1st gen) and one M5Core2.

What you think about do some play around this stuff? Is there a way to interface It with one (at least) or two nrf24 modules and run the jammer on them? Like "micro version" or "easy interfacing"... 

Im able for beta-test if you want... I'll back from hdays next 15th aug... I'll come with fresh news about the performance and range of my dual 2G4M27 jammer, tested with both antennas (omni and dir) and all jam modes... Im pretty sure It will works hard for every days... 

Thanks in advance!

W0rthlessS0ul, 2025-07-27:
Hi, it's probably possible to do this, but I don't have any m5stack devices, so I won't be able to do it anytime soon.

djmuerteitm, 2025-07-27:
Hi! Of course, your apps are always welcome, and i am very gratefull to you and for your dedication and time. If at any time you want to do some coding, all my stuff are able to test all you want.

Again, thanks for doing this. I'll write about jammer performance when come back from hdays.

closed 2025-07-27 by djmuerteitm